GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #2: Caution: Fake MAX3232 DB9 RS232 TTL converter

wavoigt, 2023-11-23:
This Converters may have Chinese Fake MAX3232 Chips, with following consequences:
Internal crosstalk on the chip leads to a distorted "echo" on the tx line which may disturb the communication with Froeling 3200!!!
see also:
[https://blog.bogpeople.com/2015/01/fake-max3232-rs232-line-drivers.html](https://blog.bogpeople.com/2015/01/fake-max3232-rs232-line-drivers.html)

I changed the MAX3232 by the original Maxim part and all worked fine.

GyroGearl00se, 2023-11-23:
Do you know any board with the genuine part where you can provide a link ?

Or do you suggest desoldering the IC from the current board and replacing it with the genuine ?

So far it seems like I haven't had any issues with my current module.

wavoigt, 2023-11-23:
Sorry, all the interface boards i found might have the fake chip.
If you have problems with the communication with any device I suggest replacing the chip.
https://www.reichelt.de/rs232-2-treiber-2-empfaenger-so-16-max-3232-cse-p57869.html?&trstct=pos_0&nbc=1

GyroGearl00se, 2023-11-23:
> Sorry, all the interface boards i found might have the fake chip.
> If you have problems with the communication with any device I suggest replacing the chip.
> https://www.reichelt.de/rs232-2-treiber-2-empfaenger-so-16-max-3232-cse-p57869.html?&trstct=pos_0&nbc=1

Ok. Thank you for the hint.
I'll keep that in mind 😃

Close this 'issue'?

wavoigt, 2023-11-23:
Thx for the ESPHome Integration of Froeling 3200 :-)
This is exactly what i've been looking for.
I'd not close the issue, so that others are aware of those chinese fake chips.

StyxyDog, 2023-12-11:
> This Converters may have Chinese Fake MAX3232 Chips, with following consequences: Internal crosstalk on the chip leads to a distorted "echo" on the tx line which may disturb the communication with [...]

Hello, I'm very interested in this project. I have a Froling PE1. I am based in Scotland and I am very nervous of connecting anything to the boiler that may cause an issue.  It is very expensive and slow to get Froling parts here from Austria! So maybe I should wait for summer before trying to get involved. 

When you say: 
> distorted "echo" on the tx line which may disturb the communication

is this an issue that might effect the proper operation of the boiler - or even potentially cause damage or is it just an issue on the HA side?

Thank you very much to @wavoigt for raising this issue and of course to @GyroGearl00se for embarking on such a project!