GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #3: Fehler in yaml Dateien?

wavoigt, 2023-11-24:
Ich hab mir mal die Mühe gemacht, die Register Adressen mit der Fröhling Tabelle zu vergleichen, und bin auf folgende Unstimmigkeiten gestossen:
(Referenz: Kommunikationsprotokoll ModBus Lambdatronic 3200 V 50.04 - B 05.17 B1200419_de | Ausgabe 07.05.2019)

1. Kessel.yaml
---------------

  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${name} Saugzug - Ansteuerung"
    id: ${name}_Saugzug_Ansteuerung
    register_type: read
#!!! adress 12 anstatt 13
    address: 12
    unit_of_measurement: "%"
    value_type: U_WORD

  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${name} Sekundärluft"
    id: ${name}_Sekundaerluft
    register_type: read
#!!! adress 13 anstatt 12
    address: 13
    unit_of_measurement: "%"
    value_type: U_WORD

2. heizkoerper.yaml
-------------------
number:
  # not correct yet ?
#!!! fehlt da nicht der register_type?
  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${name} HK1 - Gewünschte Vorlauf-temperatur bei -10°C Außentemperatur"
    id: ${name}_HK1_Vorlauf_Temperatur_minus_10C_Aussentemperatur
    address: 1032
    unit_of_measurement: "°C"
    value_type: U_WORD
    multiply: 2
    min_value: 10
    max_value: 110

GyroGearl00se, 2023-11-24:
Initially I had 12 & 13 the other way around. But then I noticed (compared with Fröling Connect App / Web Portal) that these two values were not matching but where swapped around.

Based on the documents I agree - it looks wrong :) Could you try it out and confirm how the reading looks for you ?

Number component from platform modbus_controller does not have the parameter register_type.

Here another example of number component from Boiler 01 which is working fine:
number:
  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${name} Boiler 1 Gewünschte Boilertemperatur"
    id: ${name}_Boiler_1_Gewuenschte_Boilertemperatur
    address: 1631
    unit_of_measurement: "°C"
    value_type: U_WORD
    multiply: 2
    min_value: 10
    max_value: 100

wavoigt, 2023-11-25:
Thx so much for your detailed answer.
I'll try it out asap.

closed 2023-12-12 by wavoigt