GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #4: Not working fine with ESP8266

wavoigt, 2023-12-11:
I tested the integration at 3200 and Adafruit ESP8266 Huzzah Breakout Board, does not work fine:
The wifi connection breaks after some minutes, sometimes i get some values, sometimes not.
Could it be that the Hardware Interface only works fine with ESP12?

GyroGearl00se, 2023-12-11:
Right in the beginning I started with an ESP8266. But while extending the code I ran pretty quick into - OutOfMemory exception.
That's why I moved on to ESP32. Since ever it is working fine.

You could try removing i.e. webserver, but I'm not sure if that would be enough.
Ideally you'd consider getting a board with ESP32.

wavoigt, 2023-12-12:
Yes, i tried my ESP32, and it seems to be working fine. :-)

closed 2023-12-12 by GyroGearl00se