GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #5: UART Pins ?

wavoigt, 2023-12-12:
You use UART0 (GPIO1, GPIO3) for communication with 3200.
This UART is hardwired with the USB-RS232 converter on the board, whitch may cause problems in this case.
UART0 is also used by ESPHome for Logging, and when i'm watching the serial data, i see both the readout commands for 3200 AND the logging messages.
Wouldn't it be better to use UART1 (GPIO9, GPIO10) for the 3200?

GyroGearl00se, 2023-12-12:
I used uart0 just out of convenience reasons, as the two pins are right next to each other.
Uart1 on the wemos is offset by one pin in each direction.

I just soldered one row of pins to the board 😃

To avoid conflicts I've set the baudrate to 0 for ESPHOME logging. (See line 9 in froeling.yaml)

But sure.. you can use uart1 as well with pins 9+10.

I might consider doing the same the next time and give a feedback here.
I'll be probably able to continue with this project beginning next year. The next couple of weeks are a little bit tight.

wavoigt, 2023-12-12:
Thanks so much, i appreciate your effort !

wavoigt, 2023-12-17:
I tested my ESP32 with UART2 (pins 16+17) because pins 9+10 are used for the flash memory.
Now, all is working fine!
1 thing: you run the RS232 converter with 5V, you should run it with 3.3V from ESP board because of the correct logic levels.

GyroGearl00se, 2024-01-22:
https://github.com/GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200/compare/7d0cd987a3535cbc14ca1eef222ea5998f1166fc...3adf9fb5cb31bfb3f7da55e67e511cfc505df1b3

Today I finally moved to UART2 as well :)

closed 2024-01-22 by GyroGearl00se