GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #6: Doesn't work with my P4

bcs-luggi, 2024-01-07:
Hi,

very cool project, thank you very much for it.

I tried to adapt it to my Froling P4, unfortunately with little success.
The dashboard is present, the D1 Mini is present, but none of the entities.
How to find the Problem?
Is it correct that the D1 needs power (e.g. via USB) or does it come from the P4 via the Max 3232?
- Does the P4 send?
- Data P4 --> Max 3232
- Data Max 3232 --> Mini D1?
- Data Mini D1 --> HA?

I'm a heating engineer, so please be lenient and patient :)

bcs-luggi, 2024-01-07:
Sorry, a few more info's:

modbus is on com2
protokoll 1 (RTU) and 2014
d1 mini can be reached via WiFi

wavoigt, 2024-01-07:
Do you use D1 Mini ESP32? Because it won't work fine with ESP8266! -> see closed Issue 4
Did you enable Modbus on the control panel of P4?
For the RS232, you should use UART2 (pins 16+17) -> see Issue 5
Yes, it has to be powered by USB!
Froeling.yaml (part):
`
esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

uart:
    # alternative UART2 tx_pin: (1) 17 rx_pin: (3) 16
  - id: uartfroeling
    tx_pin: 17
    rx_pin: 16
    baud_rate: 57600
    data_bits: 8
    stop_bits: 1
    parity: NONE
`
What do you see when connecting by browser to D1 mini via WiFi?
Is there a list of entities and a log output?

bcs-luggi, 2024-01-07:
Hi Walter,

thanks for your quick replies.

No ist a ESP 8266 (can't see issue 4)
Yes Modbus is enabled
I'll change it to Uart 2
I've done it with USB Power

Froeling.yaml:

esp8266:
  board: esp01_1m

Yes the list is here, but no values.

bcs-luggi, 2024-01-07:
OK, i found issue 4, sorry.

Where to get a esp32 mini without Bluetooth
I only found esp32 mini with wifi AND Bluetooth.

wavoigt, 2024-01-07:
It does not matter if it has Bluetooth.
I got it from Amazon.

bcs-luggi, 2024-01-11:
Hi,

now it's a ESP32 but no significant change.
WiFi is very stable and when I connect direkt all items are present but without any value.
I tried a lot:
- PIN 16/17
- PIN 1/3
- normal RS232 cable
- "null modem" cable

The RS232 adapter is a JZK (https://www.amazon.de/dp/B09L1BB6F8?psc=1&ref=ppx_yo2ov_dt_b_product_details)

Any idea how to check if the modbus (P4) is sending?

On the P4 I configured:

Mobus on COM 2
Modbusadress 2
Protokoll 1 (RTU)
Modus 2014

[image]

bcs-luggi, 2024-01-11:
I got this message after the last reboot while NOT connected to the P4?????

[image]

wavoigt, 2024-01-11:
Did you try to swap rx / tx wires?

wavoigt, 2024-01-11:
_"Any idea how to check if the modbus (P4) is sending?"_
if you disable Modbus in the control panel of P4, the Lambdatronic_3200 sends a status message on com2 in txt format every second along with some readings. You can check this message with a nullmodem cable connecting to PC and a terminal software like putty, Baudrate 57600, 8N1

bcs-luggi, 2024-01-12:
Hi, ok, now it works.
I've swaped tx an rx and put the RS Adapter direkt on the P4 COM2 without cable.
Great thank you very much.

GyroGearl00se, 2024-01-12:
Glad it works now for you :)
Are you sure both cables you've tested are different - straight trough and cross over (null modem) ?
The reason why I connected it with a cable is to ensure that I can mount my ESP32 outside for better WiFi signal.

Anyway... Important part is that it works now.

I've designed a case for it recently (3D Print) with the option to glue some magnets in at the bottom to make it hold on to the housing of the heating boiler. I'll upload file once I've done some optimization :)

bcs-luggi, 2024-01-13:
One of the cable's is marked as null modem, the other is marked as nothing,
so I thougt it's a normal staight trough...

Is it possible to get 3.3 V from the P4 circuit bord direkt to the 3.3 V pin on the ESP, or where do you get your power from?

wavoigt, 2024-01-13:
Not 3,3V, because it is generated on the D1 mini ESP12 board out of the Vcc +5V.
It is better using a USB Power supply.

GyroGearl00se, 2024-01-13:
I personally use as well a external power supply (USB Adapter). I didn't dare yet to pick up the power supply somewhere from the circuit board of the boiler :)

bcs-luggi, 2024-01-13:
I dont like to have another power supply.
Is it posible to put 5 V on the "VCC" to get 3.3 V on the "3V3"

GyroGearl00se, 2024-01-13:
If you can find a suitable 5V and GND on the board from your boiler unit - then you could do it (at your own risk ;⁠-⁠) ).

closed 2024-01-13 by GyroGearl00se

bcs-luggi, 2024-01-17:
Hi,

on the Board there are two 5V pin, 
- Primärluft
- Sekundärluft
It work's :)