GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #7: How to include my own yaml - Files

bcs-luggi, 2024-01-12:
Hi, it's me agin :)

As my configuartion is different form yours I have to change some values.
As far as I understand the projekt, the froeling.yaml refers to the yaml piles in your
repository here on github.

How to include my own yaml's?

I've tries with:
packages: !include_dir_named froeling

but I think the syntax is wrong.

GyroGearl00se, 2024-01-12:
if you want to have your modified files locally in your ESPHome instance you could include them like this:

packages:
  austragung: !include froeling/austragung.yaml
  boiler_01: !include froeling/boiler_01.yaml
  heizkoerper: !include froeling/heizkoerper.yaml
  kessel: !include froeling/kessel.yaml
  puffer_01: !include froeling/puffer_01.yaml
  zirkulationspumpe: !include froeling/zirkulationspumpe.yaml

Basically you can just create a subfolder in your ESPHome config directory (in this example called "froeling") and place there your personal yaml file and reference to it. No need to separate them in the categories like I've done here in this example. But I think it just gives you a better overview.

May I ask how your configuration looks like ? What components do you have ? If you're using "Fröling Connect" App, could you provide some screenshots ? I'm just curios if it would make sense to include them into this repository as well. Then everyone could just comment out maybe the yaml files where they don't have the component for it ?

bcs-luggi, 2024-01-13:
Thank's I'll try it.

My Config:
- P4
- 3 heating circuits (2 underfloor, 1 radiator)
- 2nd Boiler (stove in the living room)
- Heat Pump for hot water
- Pellets bunker with 4 sucction nozzels

When I'm finished I'l  bring it online.