GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #8: wrong display of negative outside temperature

wavoigt, 2024-01-13:
My Outside Temperature reading is currently 32762°C ;-)
you should probably change the "U_WORD" into "int" to solve this issue.

in Heizkoerper.yaml:
`  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${name} Aussentemperatur"
    id: ${name}_Aussentemperatur
    register_type: read
    address: 1000
    unit_of_measurement: "°C"
    value_type: U_WORD # change to "int"
    filters:
      - multiply: 0.5
`

bcs-luggi, 2024-01-13:
Good hint, Iwas already despairing, but he don't accept "int".
Option are: 'RAW', 'U_WORD', 'S_WORD', 'U_DWORD', 'U_DWORD_R', 'S_DWORD', 'S_DWORD_R', 'U_QWORD', 'U_QWORD_R', 'S_QWORD', 'S_QWORD_R', 'FP32', 'FP32_R'.

I'try them...

wavoigt, 2024-01-13:
So  think S_WORD would be fine.

bcs-luggi, 2024-01-13:
Yes Sir, you saved my day...

GyroGearl00se, 2024-01-13:
Thank you guys for the fix :) Didn't had the chance yet to fix it myself. Initially I just copy pasted the whole block from another sensor.
Updated in git -> https://github.com/GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200/commit/58eda89d4de89fcc4046fc2eca292e145f7107bd

closed 2024-01-13 by GyroGearl00se

wavoigt, 2024-01-14:
I still get the wrong reading.
Do i have to re-flash my ESP32 ?

GyroGearl00se, 2024-01-14:
Yes. you'd definitely need to reflash your ESP32.

As with current configuration the refresh rate is 1 day (see configuration in froeling.yaml - packages section) it could work already. Alternatively you could als clean the build files and then flash it again. This should ensure that it'll pull again the files from Github.

[image]