GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #18: values in the svg images dynamically

the-iot, 2024-01-21:
hi,

it is possible to change the values in the svg images dynamically?

the dashboard works fine but the svg images have the standard values

GyroGearl00se, 2024-01-21:
Hi @the-iot ,

could you describe your request a bit more detailed ?

The values should update with the values from your entities. Given that you haven't changed the entity names or the hostname of your ESP32 (as this would result into a different entity name than what is expected in the HA Floorplan card).

Unfortunately I'm not aware if HA Floorplan does support assigning i.e. entity "sensor.abc" to svg id "123". For now I'm only aware of that the entity name and the ID in the SVG has to match.
Was this your question ?

If anyone knowns another approach - I'd be happy to try to have a look into it.

GyroGearl00se, 2024-01-21:
What I'll consider is to replace the initial values in the svg files with blanks or "---".
Just to avoid confusion in case someone does setup the dashboard but is not reading out the data yet properly :)

the-iot, 2024-01-21:
hi,

ok... i think this is the problem. i renamed de entities because i didn't use the esp32 for my Fröling PE1.

i use a COM to Ethernet converter. 

ps.: i can explain it better in german ;)

GyroGearl00se, 2024-01-21:
Klar, kannst auch gerne auf deutsch erklären :)

Aber ich glaube ich verstehe jetzt was das Problem ist. Du nutzt eine andere Möglichkeit dir die Daten von deinem Fröling Kessel auszulesen - Würdest aber dennoch gerne diese Dashboards/Panels/SVGs verwenden.

Wie gesagt ist mir leider nicht bekannt ob "HA Floorplan" das unterstützt, dass man eine bestimmt Entität einer x beliebigen ID der SVG zuordnet. Meines Wissens müssen diese identisch sein.
Was du machen könntest (Etwas Fleißarbeit :) ). Schnapp dir die SVG Dateien aus meiner repository und editiere diese für deinen eigenen bedarf.

Du könntest entweder das z.B. mit Editor/Notepad++ machen und nach den sensor Namen suchen die ich verwendet habe und es mit deinen eigenen ersetzen. Oder... Du machst es mit einer Anwendung wie z.B. "Inkscape" (Kostenlos).

Achtung: Die Objekte in einer SVG haben eine Kennung (ID) & eine Beschriftung (inkscape label). Für HA Floorplan ist die ID relevant. Kannst natürlich aber auch beides abändern um sicher zu gehen.

Solltest du mit diesem vorgehen nicht zurecht kommen - Gerne einfach nochmal melden. Kann dich gerne Versuchen zu unterstützen das es zum laufen bekommst.

the-iot, 2024-01-21:
Super. Danke für die Erklärung. Bisher habe ich immer nur iobroker verwendet und muss mich dadurch bei HA erstmal etwas zurecht finden.

Genau ich nutze den waveshare COM to ETH PoE Modbus Converter. 

Alternativ könnte ich ja sicherlich auch die Entitäten umbennen.

Aktuell trägt die Kesseltemperatur folgende ID: sensor.modbus_pe1_furnace_temperature

Danke dir schon mal für die Hilfe und Erklärung

Gruß

GyroGearl00se, 2024-01-21:
Sehr gerne.

Ja, müsstest wie gesagt in der SVG Datei die IDs entsprechend umbenennen, so wie es eben bei dir ist. + Für die Panels/Dashboard ebenfalls :)

Einen RS232/485 to ETH converter von Waveshare habe ich tatsächlich auch zuhause. Anfänglich wollte ich das ganze hier auch damit machen. Habe mich dann aber dennoch für ESPHome entschieden.
Eher mit dem Gedanken, dass man dadurch nicht zwangsläufig an Home Assistant gebunden wäre. Mann könnte z.B. in den ESPHome code noch sehr einfach MQTT aktivieren. Dann wären zumindest die Entitäten überall integrierbar.

the-iot, 2024-01-21:
Ja das stimmt natürlich. Mit MQTT ist man etwas flexibler.

Müssen die "Rules" im Dashboard ebenfalls angepasst werden bei der SVG?

`type: custom:floorplan-card
full_height: false
config:
  image: /local/froeling/froeling_kessel.svg
  cache: false
  stylesheet: /local/froeling/home.css
  defaults:
    hover_action: hover-info
    tap_action: more-info
  rules:
    - entities:
        - **sensor.modbus_pe1_abgastemperatur**
        - **sensor.modbus_pe1_kesseltemperatur**
      tap_action: false
      state_action:
        action: call-service
        service: floorplan.text_set
        service_data: ${entity.state + "°C"}
    - entities:
        - sensor.froeling_restsauerstoffgehalt
        - sensor.froeling_puffer_1_pufferpumpen_ansteuerung
        - sensor.froeling_saugzug_ansteuerung
        - sensor.froeling_fuellstand_im_pelletsbehaelter
      tap_action: false
      state_action:
        action: call-service
        service: floorplan.text_set
        service_data: ${entity.state + "%"}
    - entities:
        - >-
          sensor.froeling_verbleibende_heizstunden_bis_zur_asche_entleeren_warnung
      tap_action: false
      state_action:
        action: call-service
        service: floorplan.text_set
        service_data: ${entity.state + "h"}
    - entities:
        - sensor.froeling_kesselzustand
      tap_action: false
      state_action:
        action: call-service
        service: floorplan.class_set
        service_data:
          class: '${(entity.state == "Heizen") ? "fill-red" : "fill-grey"}'
    - entities:
        - binary_sensor.froeling_puffer_1_pufferpumpen_ansteuerung_an_aus
      tap_action: false
      state_action:
        action: call-service
        service: floorplan.class_set
        service_data:
          class: '${(entity.state == "on") ? "fill-green" : "fill-grey"}'

Ich vermute schon oder?
`

GyroGearl00se, 2024-01-21:
Ja definitiv. Das meinte ich mit 

> "+ Für die Panels/Dashboard ebenfalls :)"

Und genau hier wäre es eigentlich ziemlich gut gewesen von HA Floorplan die Möglichkeit zu bieten eine art mapping zu haben von Entität -> ID. Aber ist wohl stand heute nicht möglich.

Ich selbst bin leider keiner der jetzt ein Dashboard mit schönen Grafiken erstellen kann welches so etwas unterstützen könnte. Deshalb habe ich zu HA Floorplan gegriffen.

GyroGearl00se, 2024-01-25:
Solltest noch fragen haben gerne nochmal melden (ggf. discussion statt issue) :)

closed 2024-01-25 by GyroGearl00se