GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #35: Error when updating esphome

wavoigt, 2024-07-04:
Hi,
I updated froeling.yaml with:
ota:
  - platform: esphome
    password: ""
Now when updating esphome i get the following error message:
[image]
my Froeling.yaml is:
[froeling.yaml.txt](https://github.com/user-attachments/files/16097973/froeling.yaml.txt)
What can i do?

GyroGearl00se, 2024-07-04:
This change is related to https://esphome.io/changelog/2024.6.0.html#ota-platforms

According to your screenshot you're already running the latest version 2024.6.6

Maybe you could try cleaning up build files ?
[image]

@Matze5593: Any ideas ?

Matze5593, 2024-07-04:
[image]
[image]

Hmmm.

I have already implemented this OTA Platform esphome, myself. but i have no problems. But my yaml is a little different.

wavoigt, 2024-07-04:
Vielen Dank, das Update hat jetzt funktioniert, "Clean Build files" war das Zauberwort.
Hätte ich auch selber drauf kommen können...
Jetzt hab ich in der ESP Konsole -> "Visit" noch 1 Warnung und 1 Fehlermeldung:

<pre>17:34:49	[D]	[sensor:094]	
'froeling Aktuelle Ansteuerung der Kollektor - Boiler Pumpe': Sending state 0.00000 h with 0 decimals of accuracy
17:34:49	[W]	[component:237]	
Component modbus_controller took a long time for an operation (52 ms).
17:34:49	[W]	[component:238]	
Components should block for at most 30 ms.
17:34:49	[D]	[text_sensor:064]	
'froeling Anlagenzustand': Sending state 'Automatik'
17:34:49	[D]	[text_sensor:064]	
'froeling Kesselzustand': Sending state 'Betriebsbereit'
17:34:49	[D]	[modbus:125]	
**Modbus error function code: 0x84 exception: 2
17:34:49	[E]	[modbus_controller:088]	
Modbus error function code: 0x4 exception: 2 
17:34:49	[E]	[modbus_controller:097]	
Modbus error - last command: function code=0x4  register address = 0x7F6B  registers count=1 payload size=0**</pre>

Aber sonst scheint alles zu funktionieren.

Matze5593, 2024-07-04:
[image]

Das kommt aus der Solarthermie yaml
Das ist der Tippfehler der "3". Es hat bei die Dateien nicht neu geladen. In der froeling yaml gibt es das refresh 1d.
Stell das mal auf 1h und mach clear build. Anschließend noch mal installieren. Dann sollte es sich die noch mal neu ziehen. Oder du wartest bist morgen und installierst dann noch mal. Hätte den selben Effekt

wavoigt, 2024-07-04:
Danke. der Tip mit dem Refresh hat geholfen.
Alles TOP :-)

closed 2024-07-06 by wavoigt