GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200 #46: svg

MaxLucien, 2025-05-05:
I've adapted the entities in both the .yaml and .svg files to my Fröling system.

However, I'm not connected via an ESP, but directly via the HA integration.

All sensors provide the corresponding values, but nothing is displayed in the svg files.

The IDs match my sensors and the yaml files.

ha-floorplan is installed using HACS.

Do you have any idea what the problem might be?

Thank you

closed 2025-05-05 by MaxLucien

Matze5593, 2025-05-05:
"However, I'm not connected via an ESP, but directly via the HA integration."

What do you mean by direct connection?

MaxLucien, 2025-05-09:
> ID: <GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200/issues/46/2851889565
> @github.com>

Ich glaube du bist Deutscher, hm ?

Meine Anlage ist online mit dem Fröling-Server verbunden und dadurch ist
auch eine Verbindung mit der Fröling-Integration in HA möglich.
Aus dieser erhalte ich meine Daten.

Gruß Michael

Matze5593, 2025-05-10:
Ja, aber welche fröling integration genau meinst du denn? Mir ist nicht bekannt das es eine gibt die auf den fröling Server oder tcp/IP zugreift.

GyroGearl00se, 2025-05-10:
@Matze5593 : vermutlich meine HA Integration die ich gebaut habe.

https://github.com/GyroGearl00se/ha_froeling_lambdatronic_modbus

@MaxLucien : Ich habe angefangen eigene Cards zu bauen womit es dann auch möglich sein soll die Entität via drop-down direkt selbst auswählen zu können. Benötigt dann auch kein HA-Floorplan mehr.
Ist noch nicht in einem sehr guten Zustand, aber ich kann demnächst gerne mal den jetzigen Stand veröffentlichen.

MaxLucien, 2025-05-11:
> ID: <GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200/issues/46/2868494106

Fröling-Connect, Gold Integration von HA.
Bekomme darüber alle Informationen der Anlage. Ich müsst aber jetzt raten
ob sich die Integration via Webportal oder direkt mit der Anlage verbindet.
Bin aber dee Meinung das es direkt ist.

Mittlerweile habe ich rausgefunden wie ich sensoren in den svg ersetzen
kann, bloß funktioniert dies nicht beständig.

Die svg zeigt nichts an, in HA ist der selbe Sensor jedoch mit korrekten
Werten ersichtlich.

Gruß

GyroGearl00se, 2025-05-13:
Ich habe mal eine custom card erstellt. Kommt ohne HA Floorplan aus und die entität kann im visuellen editor einfach konfiguriert werden.

https://github.com/GyroGearl00se/lovelace-froeling-card